stevstrong/Arduino_STM32 #12: Fail on compile

iceman1979, 2019-02-20:
This is what I'm getting when I try to compile a blink sketch. Any idea why this would not compile?

Arduino: 1.8.8 (Windows 10), Board: "Generic STM32F407V series, USB serial (CDC), STLink, Smallest (default)"

In file included from C:\Users\icema\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4/usb.h:8:0,

                 from C:\Users\icema\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/usb_serial.h:34,

                 from C:\Users\icema\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/wirish.h:50,

                 from C:\Users\icema\Documents\Arduino\hardware\Arduino_STM32\STM32F4\system/libmaple/Arduino.h:31,

                 from sketch\STM32F407VET6_BLINK.ino.cpp:1:

C:\Users\icema\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4/VCP/usb_conf.h:29:22: error: conflicting declaration 'typedef unsigned int uint32_t'

 typedef unsigned int uint32_t;

                      ^

In file included from c:\users\icema\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\lib\gcc\arm-none-eabi\4.8.3\include\stdint.h:9:0,

                 from C:\Users\icema\Documents\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/wirish.h:38,

                 from C:\Users\icema\Documents\Arduino\hardware\Arduino_STM32\STM32F4\system/libmaple/Arduino.h:31,

                 from sketch\STM32F407VET6_BLINK.ino.cpp:1:

c:\users\icema\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\stdint.h:66:20: error: 'uint32_t' has a previous declaration as 'typedef __uint32_t uint32_t'

 typedef __uint32_t uint32_t ;

                    ^

exit status 1
Error compiling for board Generic STM32F407V series.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

iceman1979, 2019-02-22:
I got it working. This is what I did to get it to work.

In usb_conf.h I added an include for stdint.h and remarked out the conflicting type declarations.

#include <stdint.h>
//typedef unsigned int	uint32_t;
typedef unsigned int	u32;
//typedef unsigned short	uint16_t;
typedef unsigned short	u16;
//typedef unsigned char	uint8_t;
typedef unsigned char	u8;

closed 2019-02-22 by iceman1979

stevstrong, 2019-05-10:
Fixed with https://github.com/stevstrong/Arduino_STM32/commit/fa683bb40cd70b1d654380515575cd1bfe954a78