stevstrong/Arduino_STM32 #13: Problem with ADC conversion F407VGT6 Blackboard

romanlife, 2019-05-02:
Hi Steve,we corresponded on the website stm32duino, I under the nickname STM32_Discoverer. So, I can't run the multichannel_nodma sketch, the led doesn't blink, the data doesn't come to the serial port. I tried to compile for the Discovery Board and for the Board Generic STM32f407V. There is no result. Сan you tell me please, BlackBoard F407VGT6 (DIYMore) refers to the Discovery F407 or generic F407 in the Arduino IDE?

stevstrong, 2019-05-03:
I tested the sketch on the [generic black F4VET6 board](http://wiki.stm32duino.com/index.php?title=STM32F407), because I don't have an F4VGT board.
The board with VGT chip onboard differs a bit from the VET board, so you have to find out where the LED is connected.
First try to blink the LED. If this works, we can go for the next step.

romanlife, 2019-05-03:
The LED onboard is PE0. After work today I will try to test the led

romanlife, 2019-05-03:
So, I compiled the sketch MultiChannel_noDMA, as before, to the serial port nothing is displayed, the led is not flashing, it is lit constantly. Then I compiled a regular sketch in which I blink the led once a second and it works, the led blinks once a second. I think that the problem is in the command- Serial.begin

stevstrong, 2019-05-03:
Do you have a schematic of the board?
Have you searched the original forum for the board and related problems?

romanlife, 2019-05-03:
[image]

romanlife, 2019-05-03:
[image]

romanlife, 2019-05-03:
[image]

romanlife, 2019-05-03:
basically I'm looking for information on the site stm32duino, but I did not find such a problem there

stevstrong, 2019-05-03:
Did you try to use serial1? This is the UART 1.

stevstrong, 2019-05-03:
And i think you need to close a jumper to Boot0 and Gnd.

stevstrong, 2019-05-03:
And i think you need a 1.5k resistor on USB Dp (D+) to 3.3v.

romanlife, 2019-05-03:
Yes Steve, on a photo a jumper boot0 is not present, but at me on a Board I already soldered it. You were right as soon as I switched to Serial1.begin everything worked, the diode flashes, data is transmitted to the serial port!!!!!!!!the sketch works.

romanlife, 2019-05-03:
I really appreciate your help, Steve

romanlife, 2019-05-04:
the problem is solved,thanks Steve

closed 2019-05-04 by romanlife

stevstrong, 2019-05-04:
Can you please tell me how did you solve the problem? It is usefull for anyone else having the same issue.

romanlife, 2019-05-04:
The problem was  that the sketch - multichannel_nodma compiles, but  but when I download it into the Board the led doesn't blink, the data doesn't come to the serial port. I used in Setup - Serial.begin(115200);But then  you recommended to try to use serial1 instead serial declaration in Setup, so I change Serial.begin(115200); - to Serial1.begin(115200); and the LED onboard flashes, data is transmitted to the serial port!!!!!!!!the sketch - multichannel_nodma works.

stevstrong, 2019-05-04:
So you did not try the 1.5k resistor on USB DP line to make USB work?

romanlife, 2019-05-04:
no, I haven't installed a resistor yet, I download sketches via ST-LINK v2 programmer, and I don't use USB yet. I think soon I will try to solder the resistor and check the performance of usb