stevstrong/Arduino_STM32 #14: Problem with DAC F407VGT6 on Blackboard

romanlife, 2019-05-07:
Hello Steve, I had a problem with the output of the DAC on F407VGT6, on the outputs PA4 and PA5, the output voltage is 1.28 V and does not change. The code of the sketch :
`
#include "libmaple/dac.h"

void setup() {
dac_init();
dac_enable(DAC_CH1);
dac_enable(DAC_CH2);
}

void loop() {
 dac_write_channel(DAC_CH1,0);
 dac_write_channel(DAC_CH2,0);
 delay(1000);
dac_write_channel(DAC_CH1,1000);
 dac_write_channel(DAC_CH2,1000);
 delay(1000);
 dac_write_channel(DAC_CH1,2000);
 dac_write_channel(DAC_CH2,2000);
 delay(1000);
 dac_write_channel(DAC_CH1,3000);
 dac_write_channel(DAC_CH2,3000);
 delay(1000);
 dac_write_channel(DAC_CH1,4000);
 dac_write_channel(DAC_CH2,4000);
 delay(1000);
}
`

stevstrong, 2019-05-07:
The DAC channel data is actually written into a buffer, not directly to the output pins.
The ADC can operate in different modes, see RM0090 chapter 14.4.
In each mode you have to set the trigger source which will cause the buffer to be output on DAC pins.
So, in your case, if you want to use SW trigger, before writing to the channels, set the trigger source to DAC_TRG_SW:
```
// set the trigger source
dac_set_trigger(DAC_CH1, DAC_TRG_SW);
dac_set_trigger(DAC_CH2, DAC_TRG_SW);
// write here the data
...
// activate trigger to write DAC data to output pins
dac_sw_trigger(DAC_CH1);
dac_sw_trigger(DAC_CH2);

```

romanlife, 2019-05-07:
I changed the code, but on the pins PA4 and PA5 is still the output voltage of 1.28 V and does not change. The code of the sketch :
`
#include "libmaple/dac.h"

void setup() {
dac_init();
// set the trigger source
dac_set_trigger(DAC_CH1, DAC_TRG_SW);
dac_set_trigger(DAC_CH2, DAC_TRG_SW);
dac_enable(DAC_CH1);
dac_enable(DAC_CH2);
}

void loop() {
dac_write_channel(DAC_CH1,1000);
dac_write_channel(DAC_CH2,1000);
// activate trigger to write DAC data to output pins
dac_sw_trigger(DAC_CH1);
dac_sw_trigger(DAC_CH2);
delay(1000);
// activate trigger to write DAC data to output pins
dac_sw_trigger(DAC_CH1);
dac_sw_trigger(DAC_CH2);
dac_write_channel(DAC_CH1,2000);
dac_write_channel(DAC_CH2,2000);
delay(1000);
}
`

stevstrong, 2019-05-07:
Try this:
```
#include "libmaple/dac.h"

void setup() {
dac_init();
dac_enable(DAC_CH1);
dac_enable(DAC_CH2);
// set the trigger source
dac_set_trigger(DAC_CH1, DAC_TRG_SW);
dac_set_trigger(DAC_CH2, DAC_TRG_SW);
}

void loop() {
// write data
dac_write_channel(DAC_CH1,1000);
dac_write_channel(DAC_CH2,1000);
// activate trigger to write DAC data to output pins
dac_sw_trigger(DAC_CH1);
dac_sw_trigger(DAC_CH2);
delay(1000);

// write data
dac_write_channel(DAC_CH1,2000);
dac_write_channel(DAC_CH2,2000);
// activate trigger to write DAC data to output pins
dac_sw_trigger(DAC_CH1);
dac_sw_trigger(DAC_CH2);
delay(1000);
}
```

romanlife, 2019-05-07:
tried your sketch, still nothing.

stevstrong, 2019-05-07:
I will check it myself in the evening.

romanlife, 2019-05-07:
Steve,I think I found the cause, sealed jumper BOOT0

romanlife, 2019-05-07:
Yep,the sketch works,thanks!!!!The problem solved

stevstrong, 2019-05-07:
I think your very first version should also work, without trigger, writing only the data.

romanlife, 2019-05-07:
yeah, I checked my first sketch, it  works too. Thanks a lot Steve

closed 2019-05-07 by romanlife