stevstrong/Arduino_STM32 #16: adc_set_sampling_time for f3 and fpu for f4

ag88, 2019-06-17:
- adc_set_sampling_time
in f3 setting the ADC sampling time calls adc_set_sample_rate()
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F3/cores/maple/libmaple/adc.c#L133
shall we rename that so that it is adc_set_sampling_time same as the f4?
actually f1 core is 'guilty' as well naming it incorrectly as adc_set_sample_rate() 

- fpu for f4
f3 turns on fpu in the default setup
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F3/cores/maple/boards_setup.cpp#L89
shall we turn the same on in F4 as well?
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/boards.cpp#L52
we can use the codes from pito which is nicely commented
http://stm32duino.com/viewtopic.php?f=39&t=2001

fpu is a nice feature of the stm32 cortex-m4, though it'd consume a little more memory
very useful if you do some dsp work like FIR filters etc.

stevstrong, 2019-06-18:
Hi, would it be possible to separate these issues? It would be more efficient to track them. Thanks

ag88, 2019-06-18:
sure thanks, i've leave the sampling time issue on this thread and created a new one for the fpu for f4.
note that on the f3 there is a minor extra indent in the fpu_enable() call that makes it looks a little less tidy, but this is simply cosmetic
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F3/cores/maple/boards_setup.cpp#L89

stevstrong, 2019-06-18:
Fixed with #16.

closed 2019-06-18 by stevstrong