stevstrong/Arduino_STM32 #17: fpu for f4

ag88, 2019-06-18:
- fpu for f4
f3 turns on fpu in the default setup
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F3/cores/maple/boards_setup.cpp#L89
shall we turn the same on in F4 as well?
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/boards.cpp#L52
we can use the codes from pito which is nicely commented
http://stm32duino.com/viewtopic.php?f=39&t=2001

fpu is a nice feature of the stm32 cortex-m4, though it'd consume a little more memory
very useful if you do some dsp work like FIR filters etc.

imho good to simply enable it as well as building with fpu is simply adding the compiler flags
-mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant
or
 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -fsingle-precision-constant
more often we may have users or even ourselves using those flags and we 'forget' to enable the fpu which requires a section of asm codes. then our sketch simply stall and we start wondering why when it is simply that missing enablefpu()

ag-123, 2019-06-19:
made a PR for this
https://github.com/stevstrong/Arduino_STM32/pull/20

ag88, 2019-06-23:
apparently this has been merged as well

closed 2019-06-23 by ag88