stevstrong/Arduino_STM32 #21: RTClock is broken

vladkozlov69, 2019-06-20:
When compiling sample, I get the compilation error:

> Compiling sketch...
"/Users/vkozlov/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -MMD -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DERROR_LED_PIN=PA7 -mcpu=cortex-m4 -DF_CPU=168000000L -DARDUINO=10805 -DVARIANT_generic_f407v -DARDUINO_ARCH_STM32F4  -DCRYSTAL_FREQ=8 -DLED_BUILTIN=PA6 -DSERIAL_USB -DUSER_ADDR_ROM=0x08000000  -mthumb -D__STM32F4__ -DSTM32F4 "-I/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/system/libmaple" "-I/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/usbF4" "-I/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple" "-I/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/variants/generic_f407v" "-I/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/src" "/var/folders/3v/mdtlw0dj3tx518cdmcm7q3lw0000gn/T/arduino_build_647884/sketch/Test_RTClock.ino.cpp" -o "/var/folders/3v/mdtlw0dj3tx518cdmcm7q3lw0000gn/T/arduino_build_647884/sketch/Test_RTClock.ino.cpp.o"
In file included from /Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/examples/Test_RTClock/Test_RTClock.ino:11:0:
/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/src/RTClock.h:98:24: error: conflicting declaration 'volatile uint32 rtc_tr'
 extern volatile uint32 rtc_tr, rtc_dr;
                        ^
In file included from /Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/src/RTClock.h:12:0,
                 from /Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/examples/Test_RTClock/Test_RTClock.ino:11:
/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/rtc.h:197:26: error: 'rtc_tr' has a previous declaration as 'volatile uint32_t rtc_tr'
 static volatile uint32_t rtc_tr, rtc_dr;
                          ^
In file included from /Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/examples/Test_RTClock/Test_RTClock.ino:11:0:
/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/src/RTClock.h:98:32: error: conflicting declaration 'volatile uint32 rtc_dr'
 extern volatile uint32 rtc_tr, rtc_dr;
                                ^
In file included from /Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/src/RTClock.h:12:0,
                 from /Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock/examples/Test_RTClock/Test_RTClock.ino:11:
/Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/rtc.h:197:34: error: 'rtc_dr' has a previous declaration as 'volatile uint32_t rtc_dr'
 static volatile uint32_t rtc_tr, rtc_dr;
                                  ^
Using library RTClock at version 1.0 in folder: /Users/vkozlov/Documents/Arduino/hardware/Arduino_STM32/STM32F4/libraries/RTClock 
exit status 1
Error compiling for board Generic STM32F407V series.

ag-123, 2019-06-20:
you may be missing an include path to your ${GCC_ARM_NONE_PATH)/arm-none-eabi/include/
try to patch up platforms.txt and add that include

ag-123, 2019-06-20:
try commenting 
line 98 in RTClock.h 
``  
//extern volatile uint32 rtc_tr, rtc_dr;
``

stevstrong, 2019-08-17:
fixed with https://github.com/stevstrong/Arduino_STM32/commit/86234b3c90b0affd57634bb46b09ec040ff92d83

closed 2019-08-17 by stevstrong

stevstrong, 2019-08-17:
Please test it and reopen this issue if there is still something wrong.

commit 86234b3c (2019-08-17): F4: RTC fix + other optimzations

- fix #21
- at_exit optimization
- gpio file renamed