stevstrong/Arduino_STM32 #27: All hardware serial and USB Serial CDC hang up in Arduino 1.8.10

xtream123, 2020-01-02:
Hi stevstrong,
Can you check on this, See link below

https://www.stm32duino.com/viewtopic.php?f=24&t=65&e=1&view=unread#unread

stevstrong, 2020-01-02:
I am investigating the issue.

stevstrong, 2020-01-03:
I just tried Serial and Serial 1, I switched back to Arduino 1.8.9. but the issue remained. 
So it seems that it is something core related and not Arduino (board or platform settings) related.

What I have observed is:
- the first bytes is transferred to the TDR register, but no TXE flag is set at the end
- the voltage level on the Tx pin is 1V, this is really strange

xtream123, 2020-01-03:
I just also want to add that in the official core (from ST) only up to Serial3 is working when I tested on it but in your core all serial is working(as I have mentioned in the forum).
I will test now in 1.8.9 using Serial and Serial1 only to see if its also happening on me.

For your observation:
1. I don't have any idea on this since I do not go very deep in the registers.
2. When does it happen?while the mcu hang? I will check on this voltage level.

xtream123, 2020-01-03:
I tried serial and serial1 in 1.8.9 and it worked for me.

the voltage level on TX pin (PA9 is 3.27V)

stevstrong, 2020-01-03:
Can you please share your working BIN file? I may have a faulty board.

xtream123, 2020-01-03:
ok I will upload it.

xtream123, 2020-01-03:
[allSerialTest.ino.txt](https://github.com/stevstrong/Arduino_STM32/files/4018564/allSerialTest.ino.txt)

Serial and Serial1 is working on that file.
Note change the file extension to ".bin"

stevstrong, 2020-01-03:
I just pushed the commit https://github.com/stevstrong/Arduino_STM32/commit/ecd8e7dd5adc019ed8f34ec36ae8cdf134d19cfc which should resolve the issue, please check out the whole repo again.

xtream123, 2020-01-03:
Thanks. I will take a look at it later and send some feedback on you.

xtream123, 2020-01-04:
hi @stevstrong , I tested your core and it worked(Serial1 to Serial6 and USB Serial at the same time) with 1.8.9 and 1.8.10 perfectly. Thank you very much.

closed 2020-01-05 by stevstrong