stevstrong/Arduino_STM32 #28: Red text appears in compiling for Arduino ide 1.8.10

xtream123, 2020-01-04:
Hi @stevstrong ,
I noticed this red text while compiling in 1.8.10 but there is no red text while compiling for 1.8.9.
Right now I don't know if there will be an impact in the mcu operation when compiled.
The sketch compiled successfully and worked(issue #27 )

Is there would be an impact in the codes?would it generate a bugs?or this is just something to be set in the core?

[1.8.10 red text during compilation_line75.txt](https://github.com/stevstrong/Arduino_STM32/files/4021670/1.8.10.red.text.during.compilation_line75.txt)

> C:\Users\MICROLOGIC SYSTEMS\Desktop\arduino-1.8.10\hardware\Arduino_STM32_stev_jan4\STM32F4\cores\maple\WString.cpp: In member function 'void String::remove(unsigned int, unsigned int)':
> C:\Users\MICROLOGIC SYSTEMS\Desktop\arduino-1.8.10\hardware\Arduino_STM32_stev_jan4\STM32F4\cores\maple\WString.cpp:700:9: warning: 'char* strncpy(char*, const char*, size_t)' accessing 0 or more bytes at offsets [0, 2147483647] and [0, 2147483647] may overlap up to 4294967295 bytes at offset [6442450941, 2147483647] [-Wrestrict]
>   strncpy(writeTo, buffer + index + count,len - index);
>   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Attached is the text file of the whole compilation process. See line 75.
Thank you.

stevstrong, 2020-01-11:
I don't think that this is an issue.
I don't get that warning, I use GCC 7.2.1.
Anyway, I found the same formula [here](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/WString.cpp#L782).

xtream123, 2020-01-11:
Okay. Thank you @stevstrong

closed 2020-01-11 by stevstrong