stevstrong/Arduino_STM32 #29: Error compiling i2c_wire_scanner example

xtream123, 2020-01-07:
Hi, 
I encountered an error compiling the i2c_wire_example
below is the error message.

```
C:\Users\LAPTOP\Desktop\arduino-1.8.9_F407_stev\hardware\Arduino_STM32_stev_jan4\STM32F4\libraries\Wire\Wire.cpp: In constructor 'TwoWire::TwoWire(uint8, uint8)':

C:\Users\LAPTOP\Desktop\arduino-1.8.9_F407_stev\hardware\Arduino_STM32_stev_jan4\STM32F4\libraries\Wire\Wire.cpp:66:20: error: 'I2C2' was not declared in this scope

         sel_hard = I2C2;

                    ^
```
Maybe its just only not declared.
I hope you can check this out.

stevstrong, 2020-01-11:
Thanks for reporting.
This [commit](https://github.com/stevstrong/Arduino_STM32/commit/77dd8c3b09c0b3e53eafa6711e48860e4cb7d529) should fix it.
Please test it and let me know if it works.

xtream123, 2020-01-11:
thanks @stevstrong , I will update you as soon as I have tested this out as I don't have my F4 board now.

I will just input something since this is related to i2c.
In your "i2c_scanner_softwire.ino" example

This
[code]
#include <SoftWire.h>

TwoWire SWire(PB6, PB7, SOFT_HSPEED);

[/code]

must be changed to this

[code]
#include <SoftWire.h>

SoftWire SWire(PB6, PB7, SOFT_HSPEED);
[/code]

Also the parameters for software i2c speed in the example must be:
in SOFT_FAST   or SOFT_STANDARD.
Since most(I guess) of the i2c device to be used is in fast or standard speed. there could be an added note to use the SOFT_HSPEED if needed by the user. This is only a suggestion

stevstrong, 2020-01-12:
It is resolved now by [this commit](https://github.com/stevstrong/Arduino_STM32/commit/637bae4d54c9d578d2ae446d5e9f309b8e308eee).

closed 2020-01-12 by stevstrong