stevstrong/Arduino_STM32 #30: changes for flash.h, flash.cpp - ART accelerator

ag-123, 2020-01-10:
hi steve,
i'm planning a commit for
STM32F4/cores/maple/libmaple/flash.h
and 
STM32F4/cores/maple/libmaple/flash.c

my plan is to introduce the same ART accelerator defines as that used in the official core
https://www.stm32duino.com/viewtopic.php?f=7&p=527#p515

i found ou, however, that the flash.h in STM32F4 core
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/libmaple/flash.h#L67
is using the definitions for stm32f103

there are 2 issues:
1   do you prefer that i update flash.h, flash.c directly or that i make a new file called flashF4.h and flashF4.c
2  this is giving me a dilemma, within F4* family there are variations
stm32F405, F407 has 3 bits in the latency field (i.e. up to 7 waits) while F401, F411 etc  has 4 bits in the latency field (15 waits)
how do we handle this, do we use definitions like __STM32F405__ , __STM32F407__, __STM32F401__, __STM32F411__, and do if defs?

ag-123, 2020-01-10:
the changes are likely done 'incremental' than big bang, i.e. i may just update FLASH_ACR definitions, and start searching for references and updating them 

no plans for other registers just yet

ag-123, 2020-01-10:
oops my goof, i found some definitions in rccF4.c
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/libmaple/rccF4.c#L138
i've been looking for them in flash.h

stevstrong, 2020-01-10:
I think we can live with 3 bits, I am not a friend of ifdefs, only if necessary.

ag-123, 2020-01-11:
i released a PR
https://github.com/stevstrong/Arduino_STM32/pull/32
there are no ifdefs ;)

the definitions are taken from the official core includes for stm32f4xx
i cross referenced f401, f411, (f405, f407, f415, f417 rm0009) and copied/adapted the definitions
pretty much all of them are there, i merged the definitions so long as they are non conflicting

ag-123, 2020-01-12:
thanks much for merging the pr !

closed 2020-01-12 by ag-123