stevstrong/Arduino_STM32 #31: Error in the SPI library

a7o, 2020-01-11:
Hi @stevstrong ,
I use your Arduino_STM32 and Ethernet_STM32 libraries with the Arduino IDE for the STM32F407VE6 and W5500 hardware. When working with W5500 via the SPI1 port, everything was OK, but I didn't succeed to use SPI2. The following code
-----
SPIClass mSpi(2);
. . .
void setup() {
. . .
  Ethernet.init(mSpi, PB12);
-----
doesn't work - the library functions continue to use SPI1 instead.
Analysing your SPIClass code I found out that it's due to the _currentSetting variable which is created as global, outside of any class, instead of being a SPIClass member. So, during the startup procedure, the mSpi instance is initialized prior to the global SPI instance, and the SPI initialization rewrites _currentSetting to point to the SPI1 port. Could you please to fix it?

Sincerely
Alex Orkin

stevstrong, 2020-01-11:
The default SPI instance is SPI1 and is defined in SPI.cpp.

If you want to use another SPI port then you do not need to define you own SPI instance.
Instead, use `SPI.setModule(2);` for SPI port 2 in setup before Ethernert.init. In this way you can switch dynamically the SPI port according to you preferences.

stevstrong, 2020-01-11:
On the other side, you are right, it should be possible to use different SPI instances without switching with `SPI.setModule()`.

closed 2020-01-11 by stevstrong

a7o, 2020-01-11:
Thank you, it really works. But there are still some problems:
1) Function
void init(SPIClass & spi, const uint8_t cs_pin)
is declared, but is practically useless and only causes misunderstanding,
2) it's impossible (or at least sophisticated a bit) to use two or more SPI ports simultaneously for different purposes (which now isn't my case though).

And now I've found yet another problem - with timers, not with SPI. In the file
Arduino_STM32_master/STM32F4/cores/maple/boards.cpp
you eliminated the function timerDefaultConfig(), putting return statement in the first line, so all timers aren't initialized and don't work.

stevstrong, 2020-01-11:
Please check out my latest commit and let me know if there are still problems.
Regarding the timers init, please open a new issue.

commit 9be674c5 (2020-01-11): F4: SPI fix #31

make _currentSetting class intern