stevstrong/Arduino_STM32 #34: Problem with timers

a7o, 2020-01-11:
I tried to get general-purpose timers working, but couldn't succeed. Analysing the library sources I've found that in the file
Arduino_STM32_master/STM32F4/cores/maple/boards.cpp
the function timerDefaultConfig() is eliminated by putting return statement in the first line. So all the timers aren't initialized and don't work.

stevstrong, 2020-01-11:
What exactly is your application and what exactly is not working?
I made that change because I had an application where there was a problem with the default configuration.

a7o, 2020-01-11:
Here's my very simple test sketch:

void setup() {

    Serial.begin(115200);
    while (!Serial) delay(10);
    delay(1000);    

    Serial.println ("Timer setup start");

    Timer2.pause();
    int period = Timer2.setPeriod(10000); // in microseconds
    Timer2.setMode(TIMER_CH1,TIMER_OUTPUT_COMPARE);
    Timer2.setOverflow(1000);
    Timer2.refresh();
    Timer2.resume();

    Serial.print ("Timer setup end, period="); Serial.println (period);
}

void loop() {
      uint16_t cnt = Timer2.getCount();
      Serial.print("timer2 counter: "); Serial.println(cnt);
      delay (500);
}

With your library, I see that the counter is always 0. After deleting the return statement, the counter changes as expected (up to 1000 and then from 0 again).

stevstrong, 2020-01-11:
I am not going to enable that line in `boards.cpp` again.
But I just pushed a commit to solve the issue.

In order to use any timer from now on, one should call
```
TimerX.init();
```
This way the timer will be initialized and paused at the same time.

So in your sketch you have to replace 
`Timer2.pause();`
with
`Timer2.init();`
Please try this and let me know if it works.

stevstrong, 2020-01-11:
I just tested your sketch modified as suggested and it works, so I will close this now.
Reopen if it does not work for you.

closed 2020-01-11 by stevstrong