stevstrong/Arduino_STM32 #35: rcc clock settings codes may need to be reworked due to different crystals

ag88, 2020-01-20:
hi steve,

i think we need to rework the clock setting f4 code.
blackpill f411 runs on a 25mhz crystal
while
disco f411 runs on a 8mhz crystal
https://www.stm32duino.com/viewtopic.php?p=714#p714

Bingo600, 2020-01-20:
Steve i just posted a git diff , for getting 84/96MHz on a F4 , autodetecting if HSE is 8 or 25MHz , same stuff as was in for 168MHz,

Find it here
https://www.stm32duino.com/viewtopic.php?f=64&t=120

/Bingo

stevstrong, 2020-02-03:
I just reworked the RCC clock settings, see https://github.com/stevstrong/Arduino_STM32/commit/1fa1a216c809e4d104cecd350258498ff878f1d9.
Changes:
- PLL_M is always set to `CRYSTAL_FREQ` (crystal frequency in MHz)
- the system clock is chosen so that USB is always supported.

For custom clock dividers one can define `USE_CUSTOM_RCC_CLK` and set the custom values in `clock_config` array.
Please reopen if you find an issue with these changes.

closed 2020-02-03 by stevstrong