stevstrong/Arduino_STM32 #38: Timers 9-14 interrupts don't work

a7o, 2020-03-15:
Hi
I work with STM32F407. Trying to set update interrupt handler for the timer 12, I found out that it actually works only for timers from 1 to 8. For the timers 9-14 the function enable_nonmuxed_irq() (module timer.c), which is called from HardwareTimer::attachInterrupt(), does nothing. Actually these interrupt vectors are shared with another hardware, so when I wrote
`Timer8.attachInterrupt(TIMER_BREAK_INTERRUPT, myTimer12Handler);`
instead of 
`Timer12.attachInterrupt(TIMER_UPDATE_INTERRUPT, myTimer12Handler);`
I got it work.

stevstrong, 2020-03-15:
Can you please post a simple example to reproduce the problem?

stevstrong, 2020-03-15:
Ok, I see your point, I will prepare a fix.

stevstrong, 2020-03-15:
Timer 1 and 8 BRK, UP and TRG interrupt vectors are shared with Timer9-14 global interrupts.
This commit https://github.com/stevstrong/Arduino_STM32/commit/b35fa41567a8be84db9467d24dbaff8c9ed64eb4 should fix it.
Please test it and let me know the result.
You may close the issue if it is solved.

a7o, 2020-03-15:
No. It still doesn't work.
Here's my simple test:
```
// Test Timer12 interrupts

static uint32_t intCounter = 0;
static timer_gen_reg_map *tmregs;
void myTimerHandler(void);
void timerSetup(void);

void setup ()
{
    Serial.begin(115200);
    delay(2000); 

    timerSetup();

    Serial.println ("Setup done");
}

void loop()
{
      Serial.print ("Intr counter = "); Serial.print (intCounter);
      Serial.println();
      delay (3000);
}

//----------------

#define  myTimer Timer12

void timerSetup(void)
{
      // timer registers base address
  tmregs = myTimer.c_dev()->regs.gen;

  myTimer.init();
  myTimer.pause();

  tmregs->CR1 = 0
          ;
  tmregs->SMCR = 0                  // Slave mode disabled
          ;
  tmregs->DIER = 0
          | TIMER_DIER_UIE            // update interrupt enabled
          ;
  tmregs->CCMR1 = 0
          ;
  tmregs->PSC = 0                   // input frequency prescaler (factor - 1)
          | (((myTimer.getClockSpeed() / 1000000) / 42) - 1)
          ;
  tmregs->ARR = 
          41999               // period = 1 ms
          ;
  tmregs->CCR1 = tmregs->CCR2 = 0
          ;
  tmregs->CCER = 0
          ;
  tmregs->EGR |= TIMER_EGR_UG       // update registers
          ;
  tmregs->CR1 |= TIMER_CR1_CEN
          ;

//	Works with Timers1...8, but doesn't work with Timers9...14
//  myTimer.attachInterrupt(TIMER_UPDATE_INTERRUPT, myTimerHandler);

//	Works fine with Timer12
  Timer8.attachInterrupt(TIMER_BREAK_INTERRUPT, myTimerHandler);
}

void myTimerHandler(void)
{
  tmregs->SR = 0;
  ++intCounter;
}
```
It seems me that the cause is here:
```
static void enable_nonmuxed_irq(const timer_dev *dev) {
    switch (dev->clk_id) {
    case RCC_TIMER2:
        nvic_irq_enable(NVIC_TIMER2);
        break;
    case RCC_TIMER3:
        nvic_irq_enable(NVIC_TIMER3);
        break;
    case RCC_TIMER4:
        nvic_irq_enable(NVIC_TIMER4);
        break;
    case RCC_TIMER5:
        nvic_irq_enable(NVIC_TIMER5);
        break;
    case RCC_TIMER6:
        nvic_irq_enable(NVIC_TIMER6_DAC);
        break;
    case RCC_TIMER7:
        nvic_irq_enable(NVIC_TIMER7);
        break;
    default:   //!!!!!!!
        ASSERT_FAULT(0);
        break;
    }
} 
```

stevstrong, 2020-03-16:
I see what you mean, the NVIC interrupts are still not enabled for Timers 9-14.
I will prepare another PR soon.

stevstrong, 2020-03-16:
The last two commits should have solved the issue, I have tested Timer11 and it worked:
#define  myTimer Timer11

```
void timerSetup(void)
{
  myTimer.init();
  myTimer.setPrescaleFactor(CLOCK_SPEED_MHZ);
  myTimer.setOverflow(50000);
  myTimer.refresh();
  myTimer.attachInterrupt(TIMER_UPDATE_INTERRUPT, myTimerHandler);
  myTimer.resume();
}
```
Please test it again.

a7o, 2020-03-16:
Thanks, now it seems to be OK, except that my antivirus treats the file wdi-simple.exe as infected with Trojan.MulDrop11.16603.

stevstrong, 2020-03-16:
Sorry, but that is not relevant for this issue, right?
I never had problems using that exe.

a7o, 2020-03-16:
Surely. I only meant to warn you about possible danger, as a precaution.

closed 2020-03-16 by a7o