stevstrong/Arduino_STM32 #39: timers not clocked at init

ag88, 2020-05-08:
hi while messing with the timers on F4. i found it strange that the timers registers did not retain their values. on tracing the codes in debug.
i found that timers are not initialized.
it turns out there is a return statement right at the beginning of the timers initialization.
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/boards.cpp#L125
any reasons for this codes, effectively by passing initialization?

stevstrong, 2020-05-08:
I had an application in which I had headache figuring out why it did not work.
The reason was [line 134](https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/boards.cpp#L134) which sets the prescaler. That is why I decided to disable the whole timer initialization part.
In user applications the timer should anyway be paused, configured and resumed. One more line to initialize it at the beginning must be added in the future.

ag88, 2020-05-08:
i'm thinking for timer initialization, it is adequate to simply clock the timers and pause() (disable) the timer. my guess is that this should reduce power consumption somewhat since the timers are not running.

i'm also thinking if we should try to be 'helpful' (e.g. setting TIMER_CR1_ARPE) or that we'd just leave it at the reset state but just disable the timers.

The initialization are useful but that it should probably be done in user codes since we'd not know ahead the application scenario that is relevant. i'd think timers probably can have more support functions in the library e.g. output control, the thing is sometimes timers are just used to run other peripherals and the user may not want output to go on the pins. But we can work those additional functions as we go along.

commit cf6a037f (2020-05-09): F4: fixes #39
closed 2020-05-09 by stevstrong