stevstrong/Arduino_STM32 #40: F4 possible bug in pheriperials bid-band macro

ag88, 2020-05-08:
while tracing some codes in debug, i noted this
https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/cores/maple/libmaple/bitband.h#L44
```
#define __bb_addr(addr, bit, bb_base, bb_ref) \
	(volatile uint32*)((bb_base) + (((uint32)((uint32)(address) - (bb_ref))) * 32) + ((bit) * 4))
```
i think that addr argument should read address to be in line with the macro, e.g.
```
#define __bb_addr(address, bit, bb_base, bb_ref) \
	(volatile uint32*)((bb_base) + (((uint32)((uint32)(address) - (bb_ref))) * 32) + ((bit) * 4))
```

stevstrong, 2020-05-08:
Hm, weird that this did not come up so far. The compiler should report an error here at each build time.
How did you spot this?

ag88, 2020-05-08:
well, i'm tracing some codes as the (timer) isr didn't fire. so i dig into the 'attachInterrupt()' codes, and more commonly, they enable the interrupts (flags) as well. those flag setting codes calls
_bb_perip_ and lead me to ___bb_addr_ macro. Hence, i spotted what i think may be a bug.

stevstrong, 2020-05-08:
Just committed the fix.
Thanks for reporting it.

closed 2020-05-08 by stevstrong

ag88, 2020-05-08:
thanks for the follow up !