stevstrong/Arduino_STM32 #41: updates in ADC.h and ADC.h

ag88, 2020-05-21:
hi, this is a heads up for now

i'm making a usb oscilloscope out of a F401 pill board, after working out the various bugs, issues.It has reached stability. The app seem to run in all cases , even though i tried to 'abuse' my app. so the additions should be rather robust. i've also reviewed the ref manual for F405/F407 and see that they are after all the same definitions (in fact i work with F407 manual and back checked in F401), so it should work across the series.F405/F407 actually have more register bits as F401 but F401 is a subset of F405/F407/F411

i'd like to commit my changes back to the F4 core.  I'm still in the process of tidying up my codes, preparing to publish my project and hence i've not made any PR or commits to the github repositories yet.

some questions are, would you prefer that i commit it (the PR) on your repository or roger's repository? i'm thinking that committing it with your repository firsthand may be 'better' in a sense that if you have made other changes, the changes can be merged firsthand so that the subsequent PR to roger's core can be merged together.

the changes done are:
- adc_prescaler  
this is modeled after the same function in the F1 core, the macro symbols are same as well. this would make it more 'compatible' for those writing codes that may possibly be used in both F1 and F4 sets.

adc.h
```
/**
 * @brief STM32F1/F4 ADC prescalers, as divisors of PCLK2.
 */
typedef enum adc_prescaler {
    /** PCLK2 divided by 2 */
    ADC_PRE_PCLK2_DIV_2 = 0,
    /** PCLK2 divided by 4 */
    ADC_PRE_PCLK2_DIV_4 = 1,
    /** PCLK2 divided by 6 */
    ADC_PRE_PCLK2_DIV_6 = 2,
    /** PCLK2 divided by 8 */
    ADC_PRE_PCLK2_DIV_8 = 3,
} adc_prescaler;

void adc_set_prescaler(adc_prescaler pre);
```
adc.c
```
/*
 *  @param pre, setup the ADC prescaler to one of PCLK /2, /4, /6, /8
 *         the templates per adc_prescaler enum
 */
void adc_set_prescaler(adc_prescaler pre) {

	ADC_COMMON->CCR &= ~ ADC_CCR_ADCPRE;
	ADC_COMMON->CCR |= ((pre & 3U) << ADC_CCR_ADCPRE_SHIFT);
}
```

- add static inline void adc_awd_disable()

adc.h
```
static inline void adc_awd_disable(const adc_dev * dev)
{
	dev->regs->CR1 &= ~ADC_CR1_AWDEN;
}
```

- add ADC overrun interrupt support

adc.h
```
//Added by bubulindo - Interrupt ID's for ADC
typedef enum {
    ADC_EOC,     // End Of Conversion interrupt.
    ADC_AWD ,    // Analog WatchDog interrupt
    ADC_JEOC,    // Injected End Of Conversion interrupt.
    ADC_OVR,	 // overrun interrupt
    ADC_LAST_IRQ_ID
} adc_irq_id;

void adc_ovr_enable_irq(const adc_dev* dev);
```

adc.c
```
void adc_ovr_enable_irq(const adc_dev* dev)
{
    dev->regs->CR1 |= ADC_CR1_OVRIE;
    //nvic_irq_enable(NVIC_ADC_1_2_3);
}

void adc_attach_interrupt(const adc_dev *dev,
                            adc_irq_id irq_id,
                            voidFuncPtr handler)
{
    if (irq_id<ADC_LAST_IRQ_ID)
    {
        (*(dev->handler_p))[irq_id] = handler;
        if(irq_id == ADC_EOC)
        	adc_enable_irq(dev);
        else if (irq_id == ADC_AWD)
        	adc_awd_enable_irq(dev);
        else if (irq_id == ADC_OVR)
        	adc_ovr_enable_irq(dev);
    }
}

/*
    ADC IRQ handler. 
    added by bubulindo
*/
void adc_irq_handler(const adc_dev * dev)
{
    //End Of Conversion
    if (dev->regs->SR & ADC_SR_EOC) {
        dev->regs->SR = ~ADC_SR_EOC;
        voidFuncPtr handler = (*(dev->handler_p))[ADC_EOC];
        if (handler) {
            handler();
        }
    }
    //Injected End Of Conversion
    if (dev->regs->SR & ADC_SR_JEOC) {
        dev->regs->SR = ~ADC_SR_JEOC;
        voidFuncPtr handler = (*(dev->handler_p))[ADC_JEOC];
        if (handler) {
            handler();
        }
    }
    //Analog Watchdog
    if (dev->regs->SR & ADC_SR_AWD) {
        dev->regs->SR = ~ADC_SR_AWD;
        voidFuncPtr handler = (*(dev->handler_p))[ADC_AWD];
        if (handler) {
            handler();
        }
    }

    //ADC overrun interrupt
    if (dev->regs->SR & ADC_SR_OVR) {
        dev->regs->SR = ~ADC_SR_OVR;
        voidFuncPtr handler = (*(dev->handler_p))[ADC_OVR];
        if (handler) {
            handler();
        }
    }

}
```

stevstrong, 2020-05-22:
You can post the PR for Roger's core, I will pull it from there.

ag88, 2020-05-22:
hi steve. thanks for your response ! :)

closed 2020-05-22 by ag88