stevstrong/Arduino_STM32 #42: Unable to compile blink project when board Maple(RET6) is select from Arduino studio

yuanb, 2020-07-19:
Hi, 

I hope this is the right place to log this issue, if not please close this issue and tell me where to log it

Environment:
OS: Mac OS X
Arduino Studio 1.18.12
STM32duino: 2020.6.20
Board: Leaflabs Maple RET6

Please check attached image for details. The same code compiles fine when Maple Rev3 is selected.
My remedy is to add the following to the beginning of Arduino15/packages/stm32duino/hardware/STM32F1/2020.6.20/cores/maple/sdio.cpp. 

Note, if I uncomment the #ifdef   __cplusplus block, the code still wont compile, same error.

//#ifdef __cplusplus
//	#include <algorithm>
//	using namespace std;
//#else // C
	#include <stdlib.h>
	#ifndef min
		#define min(a,b) ((a)<(b)?(a):(b))
	#endif // min

	#ifndef max
		#define max(a,b) ((a)>(b)?(a):(b))
	#endif // max
//#endif // __cplusplus

[image]

stevstrong, 2020-07-21:
OK, I will have a look on this.

stevstrong, 2020-08-03:
Your core files are from: http://dan.drown.org/stm32duino/package_STM32duino_index.json
This error is related to the Roger's core: https://github.com/rogerclarkmelbourne/Arduino_STM32
I think this is fixed by this commit: https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/58ca12ab5bf7e844733897fbfb4cad790af1a50b
Please check out the new file sdio.cpp and try to compile again.

yuanb, 2020-08-04:
@stevstrong  I don't understand how Arduino studio manage its packages. Are you saying I'm actually not using your code?

stevstrong, 2020-08-04:
Yes, that is what I am saying.
If you want to use my core files, then you have to install it manually, identically as described here: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation
just that the package you have to download is this one: https://github.com/stevstrong/Arduino_STM32/archive/master.zip

yuanb, 2020-08-05:
Thanks for letting me know, I forgot how my package was installed. I will try yours

stevstrong, 2020-08-06:
It is good that it works with Roger's core, use that, you don't need to use my core.
Can you please  post the error messages as text?

stevstrong, 2020-08-10:
I will close this as it seems to be resolved.

closed 2020-08-10 by stevstrong