stevstrong/Arduino_STM32 #44: STM32F401, F405, F407, f4xx? backup registers

ag88, 2021-03-02:
currently backup registers in the F4 libmaple core don't seem to be reflecting the F4 backup registers
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/cores/maple/libmaple/bkp.h#L101

while attempting to use them, i ran into some bummers. the base address seemed inappropriate. Hence, i've made some changes in my local copy of bkp.h and bkp.c. Apparently on f401, f405, f407, f427 there are 20 backup registers. However, i'm not too sure about the rest.

i've posted a discussion thread in the forum
https://www.stm32duino.com/viewtopic.php?f=63&t=947

I can make a PR if you prefer. However, i've not done that as i'm not sure if the F4xx family has only 20 RTC backup registers as do f401, f405, f407, f427, these are the ones i've just browsed the datasheets and manuals.

my own code tweaks works for myself, but i'm just raising it in case it is useful.

stevstrong, 2021-03-03:
Hm, the core is using the value according to STM32F405/407 manual (DocID022152 Rev 8) page 73, the BKPSRAM address of 0x4002 4000.

So what exactly do you mean that is not correct?

stevstrong, 2021-03-03:
Ah, I see what you mean, I entered the SRAM address instead of RTC & backup register address which is 0x4000 2800.

ag88, 2021-03-03:
hi steve, 

i've uploades codes for my local copy of bkp.h and bkp.c in a gist
https://gist.github.com/ag88/c4429b0e77b58e2b3d461798d6021dd5

what is missing here is a test or more appropriately example sketch to use it. After reviewing the various specs and manual.
i think it is fairly safe to conclude that stm32 f4xx has 20 backup registers all 32 bits and for the larger chips they have addition of 4KB backup sram. the specs i've reviewed are only 
(only 20 backup registers) f401, f411,
(20 backup registers + 4kb sram) f405, f407, f427, f429, f439)
quite limited really, but that's what observed.

in my use case, i'm using the backup registers on stm32f401 with RTC, i.e. RTC is initialised on HSE and the registers are write enabled. so i'm not too sure if bkp.h, bkp.c alone is adequate to use the backup registers. it would take me a little time to create the test/example sketch and i can create that in a PR for roger's core if you prefer.

stevstrong, 2021-03-03:
It looks ok, you can push a PR to Roger's core, I will then check it.

ag88, 2021-03-03:
edit: reading a little further in [rm0009](https://www.st.com/resource/en/reference_manual/dm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf)

under section 2.3 Memory map , BKPSRAM is at 0x4002 4000 
then in section 5.1.2 battery backup domain - backup domain access
" 
• Access to the RTC and RTC backup registers
1. Enable the power interface clock by setting the PWREN bits in the RCC_APB1ENR
register (see Section 7.3.13 and Section 6.3.13)
2. Set the DBP bit in the Section 5.4.1 and PWR power control register (PWR_CR) for STM32F42xxx and STM32F43xxx to enable access to the backup domain
3. Select the RTC clock source: see Section 7.2.8: RTC/AWU clock
4. Enable the RTC clock by programming the RTCEN [15] bit in the Section 7.3.20: RCC Backup domain control register (RCC_BDCR)

• Access to the backup SRAM
1. Enable the power interface clock by setting the PWREN bits in the RCC_APB1ENR
register (see Section 7.3.13 and Section 6.3.13 for STM32F405xx/07xx and
STM32F415xx/17xx and STM32F42xxx and STM32F43xxx, respectively)
2. Set the DBP bit in the PWR power control register (PWR_CR) for STM32F405xx/07xx
and STM32F415xx/17xx and PWR power control register (PWR_CR) for
STM32F42xxx and STM32F43xxx to enable access to the backup domain
3. Enable the backup SRAM clock by setting BKPSRAMEN bit in the RCC AHB1
peripheral clock enable register (RCC_AHB1ENR)."

I'd guess we can enable that in bkp_init(). step 1 and 2 seem to be common.
we can add step 3 for the sram part. 

But for the backup registers, apparently users would need to initialise rtc (e.g. rtc.begin(). it so happens that in my case rtc is running. note that i've also customized my copy of RTClock.cpp. rtc.begin() checks if LSE_ON is set. if LSE_ON is set (i.e. LSE running), i skipped backup domain reset and resetting the time and date. This is to allow rtc.begin() to be called on power up and that VBAT is maintained by a coin cell but it should not reset/clear the existing running rtc.

and we can perhaps define a macro in bkp.h
`` #define BKPSRAM_BASE          0x40024000UL // Backup SRAM(4 KB) base address  ``
this define is found in the cmsis header in the official core for stm32f407
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f407xx.h

keeping the 'old' somewhat compatible libmaple backup register bkp.h functions is useful as it would enable apps to be written for both libmaple f1 and f4 cores without if-defs, even if at hardware level it is literally different. stm32f103c8 is a 'subset' of stm32f4

ag88, 2021-03-03:
> It looks ok, you can push a PR to Roger's core, I will then check it.

thanks, i'd do so in the next few days, note there is no hurry with this whatsoever. 
i just happen to start using this and i just thought i'd 'tidy up' some of the use cases related to bkp and rtc

stevstrong, 2021-03-04:
I reopen the issue until we adapt the backup functionality as you indicated above.

ag88, 2021-03-08:
just a little update on progress. Currently it is still 'work in progress', a sample run on stm32f401cc black pill looks like this.
this is just for the backup registers.
```
init
bkp values:
reg :1:0
reg :2:0
reg :3:0
reg :4:0
reg :5:0
reg :6:0
reg :7:0
reg :8:0
reg :9:0
reg :10:0
reg :11:0
reg :12:0
reg :13:0
reg :14:0
reg :15:0
reg :16:0
reg :17:0
reg :18:0
reg :19:0
reg :20:0
setting bkp values:
bkp values:
reg :1:100001
reg :2:100002
reg :3:100003
reg :4:100004
reg :5:100005
reg :6:100006
reg :7:100007
reg :8:100008
reg :9:100009
reg :10:100010
reg :11:100011
reg :12:100012
reg :13:100013
reg :14:100014
reg :15:100015
reg :16:100016
reg :17:100017
reg :18:100018
reg :19:100019
reg :20:100020
```
i've verified that they persist across rest and remains there if vbat is maintained. RTC is initialised and running on LSE

ag88, 2021-03-10:
ok the PR is done
https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/850
works on stm32f407ve black board (bkp registers and bkp sram) and stm32f401 (bkp registers)
sample run
```
init
bkp values:
reg :1:100001
reg :2:100002
...
reg :19:100019
reg :20:100020
BKPSRAM 2nd byte:63
BKPSRAM 2nd uint16 word:10676
BKPSRAM 2nd uint32 word:3105925270
BKPSRAM text at offset 10:s@▒▒+1▒▒▒▒▒
A▒▒vu@▒r▒v▒|▒/▒P7W%▒O▒▒
setting bkp values:
bkp values:
reg :1:100001
reg :2:100002
...
reg :19:100019
reg :20:100020
BKPSRAM 2nd byte:100
BKPSRAM 2nd uint16 word:10000
BKPSRAM 2nd uint32 word:1000000
BKPSRAM text at offset 10:a quick brown fox jumps over the lazy dog
```

after reset looks like this
```
init
bkp values:
reg :1:100001
reg :2:100002
...
reg :19:100019
reg :20:100020
BKPSRAM 2nd byte:100
BKPSRAM 2nd uint16 word:10000
BKPSRAM 2nd uint32 word:1000000
BKPSRAM text at offset 10:a quick brown fox jumps over the lazy dog
setting bkp values:
bkp values:
reg :1:100001
reg :2:100002
...
reg :19:100019
reg :20:100020
BKPSRAM 2nd byte:100
BKPSRAM 2nd uint16 word:10000
BKPSRAM 2nd uint32 word:1000000
BKPSRAM text at offset 10:a quick brown fox jumps over the lazy dog
```

ag88, 2021-03-11:
one more thing, i've made some changes in RTClock.cpp.
this is so that rtc.begin() would not be overwritten if it is running on LSE and VBAT.
https://gist.github.com/ag88/c4429b0e77b58e2b3d461798d6021dd5#file-rtclock-cpp
I've tested them several rounds and the changes are stable on stm32f407ve black and stm32f401cc black pill.
it is a useful update. can i include them in this PR? at the moment i've only updated codes in my gist so that it can be reviewed

stevstrong, 2021-03-11:
Feel free to include it in the PR.

ag88, 2021-03-11:
thanks, i've added the commit.
note that i've modified Test_RTClock example during my regression tests.
https://gist.github.com/ag88/c4429b0e77b58e2b3d461798d6021dd5#file-test_rtclock-cpp
among the changes 
it seemed rtc.begin() is not called in setup(). that is necessary or normally freezes / hardfaults would happen.
i've made various other edits for the interactivity changes and commented codes changing the weekday, for some reason i've not figured out, the weekday changing codes tend to clobber a RTC that is already running ok.
the example is not in the commit and only in my gist in the above link

stevstrong, 2024-02-29:
The changes were merged from the upstream repository.

closed 2024-02-29 by stevstrong